seud0nym/sigenergy2mqtt #9: Error setting grid export limit

lesson07quirky, 2025-07-19:
I’m seeing the following errors in the log when trying to set grid export limit to 0.3kWh. Also tried 1kWh with the same result.

```
2025-07-19 15:18:34,806 INFO  base:906 GridMaxExportLimit - write_registers self._address=40038 value=300.0 (self.latest_raw_state=500) slave=247
2025-07-19 15:18:34,807 ERROR base:949 GridMaxExportLimit write_registers threw ModbusClientMixin.write_registers() takes 3 positional arguments but 4 positional arguments (and 1 keyword-only argument) were given
2025-07-19 15:18:34,808 WARNING base:1158 GridMaxExportLimit - Attempt to set value to 0.3 FAILED: ModbusClientMixin.write_registers() takes 3 positional arguments but 4 positional arguments (and 1 keyword-only argument) were given
2025-07-19 15:20:26,680 INFO  base:906 GridMaxExportLimit - write_registers self._address=40038 value=300.0 (self.latest_raw_state=500) slave=247
2025-07-19 15:20:26,681 ERROR base:949 GridMaxExportLimit write_registers threw ModbusClientMixin.write_registers() takes 3 positional arguments but 4 positional arguments (and 1 keyword-only argument) were given
2025-07-19 15:20:26,682 WARNING base:1158 GridMaxExportLimit - Attempt to set value to 0.3 FAILED: ModbusClientMixin.write_registers() takes 3 positional arguments but 4 positional arguments (and 1 keyword-only argument) were given
2025-07-19 15:21:07,593 INFO  base:906 GridMaxExportLimit - write_registers self._address=40038 value=300.0 (self.latest_raw_state=500) slave=247
```

lesson07quirky, 2025-07-19:
Nice one, thanks 😊

What triggers the ha addon update? I can’t seem to get mine to update, not sure if it’s something I’m doing or not. I’ve tried the usual update methods in the UI and also a `ha addons reload` in the terminal but it’s suggesting 2025.7.13-1 is the latest.

seud0nym, 2025-07-19:
Sorry, but I found an issue when testing the add-on and I didn't have time to fix it last night, so it will be published sometime this morning.

seud0nym, 2025-07-19:
It has just finished uploading, so if you do an update through the UI, it should be there now.