seud0nym/sigenergy2mqtt #14: Unable to set use-simplified-topics with HA Add-On

sam-ward, 2025-08-02:
I am trying to use a custom sigenergy2mqtt.yaml config file with my home assistant add-on instance of sigenergy2mqtt so that I can enable the `use-simplified-topics` feature as I also want to consume the data elsewhere from the mqtt broker. However the add-on won't start and is throwing the following error:
```
2025-08-02 08:16:14,415 INFO  __init__:23 Release 2025.7.20
2025-08-02 08:16:14,429 INFO  config:44 Loading configuration from /config/sigenergy2mqtt.yaml...
2025-08-02 08:16:14,440 CRITICAL __init__:451 Error processing configuration: home-assistant.configuration element contains unknown option 'use-simplified-topics'
[00:16:14] WARNING: Halt add-on with exit code 1
```

My sigenergy2mqtt.yaml file is as follows:
```
home-assistant:
  # enabled
  #   default: false
  #   description:  If true, sigenergy2mqtt will publish auto-discovery 
  #                 messages for Home Assistant to automatically configure
  #                 the devices and entities.
  enabled: true
  # use-simplified-topics
  #   default: false
  #   description:  If true, sigenergy2mqtt will use a simplified topic
  #                 structure for Home Assistant entities. The topic will be
  #                 sigenergy2mqtt/object_id/state instead of the full Home
  #                 Assistant topic structure of 
  #                 homeassistant/platform/device_id/object_id/state.
  use-simplified-topics: true
```

Is this an option that should be available to use in the add-on?

Cheers

seud0nym, 2025-08-02:
The `use-simplified-topics` option was added in Release 2025.7.26, but the Home Assistant add-on is still using Release 2025.7.20.

I am just in the process of finalising the next release which will be at parity across all platforms. Hopefully it will be today if I don't encounter any issues.

seud0nym, 2025-08-02:
I have just published Release 2025.8.2 which should fix this.

seud0nym, 2025-08-02:
There is an embarrassing bug (#15) in 2025.8.2 which is fixed in 2025.8.2-1.

sam-ward, 2025-08-02:
Just forced a refresh and have updated the add-on, and the new release fixes the SIGENERGY2MQTT_CONFIG KeyError.

With the config file in place, I now have a different error:
```RITICAL mqtt:96 Connection to MQTT broker core-mosquitto REFUSED - Not authorized```

Which i'm interpreting to mean that there are no MQTT credentials being supplied. 

My question is, when I first ran the add-on it obtained credentials automatically (I presume from the HA supervisor), which is all good. When I specified the config file though, it doesnt seem to get credentials automatically any more.  But in the add-on documentation it says:
```NOTE: If you provide both an advanced configuration file and enter options into the Configuration tab, the Configuration tab will override any identical settings in the configuration file. If you are using the Mosquitto Broker addon, the MQTT host, port, username and password will always override what is the configuration file. ```

So I am not sure whether there is a bug here, or I am just misunderstanding the doco / the doco might need to be clarified slightly.

seud0nym, 2025-08-03:
Yes, it is a bug. Unfortunately, my test configuration file contained credentials for an external MQTT server, which obscured the problem.

I have fixed the merging of the add-on configuration with the configuration file, and published a new version of the add-on (2025.8.3).

seud0nym, 2025-08-11:
I believe this should be fixed, but if not, please re-open with any additional information.

closed 2025-08-11 by seud0nym