seud0nym/sigenergy2mqtt #15: KeyError: 'SIGENERGY2MQTT_HASS_ENABLED'

boma304, 2025-08-02:
The addon stops with KeyError: 'SIGENERGY2MQTT_HASS_ENABLED' . I'm not using a configuration file.
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun sigenergy2mqtt (no readiness notification)
s6-rc: info: service legacy-services successfully started
[09:54:22] INFO: Configuring...
[09:54:22] INFO: No configuration file found, using default configuration
[09:54:23] INFO: MQTT service available, using host: core-mosquitto, port: 1883, username: addons
[09:54:23] INFO: Checking for MQTT override configuration...
[09:54:23] INFO: MQTT configured, using username: addons
[09:54:23] INFO: MQTT configured, using host: core-mosquitto, port: 1883, username: addons
[09:54:24] INFO: Read-only mode enabled, no write operations can be performed.
[09:54:25] INFO: Metrics disabled, not publishing metrics to Home Assistant.
[09:54:25] INFO: Starting sigenergy2mqtt service...
[09:54:25] INFO:  -> Parameter: --hass-enabled
[09:54:25] INFO:  -> Parameter: --mqtt-broker=core-mosquitto
[09:54:25] INFO:  -> Parameter: --mqtt-port=1883
[09:54:25] INFO:  -> Parameter: --mqtt-username=addons
[09:54:25] INFO:  -> Parameter: --mqtt-password=[REDACTED]
[09:54:25] INFO:  -> Parameter: --modbus-host=192.168.1.13
[09:54:25] INFO:  -> Parameter: --modbus-inverter-device-id
[09:54:25] INFO:  -> Parameter: 1
[09:54:25] INFO:  -> Parameter: --modbus-readonly
[09:54:25] INFO:  -> Parameter: --no-metrics
2025-08-02 09:54:25,732 INFO  __init__:24 Release 2025.8.2
Traceback (most recent call last):
  File "/root/.local/bin/sigenergy2mqtt", line 3, in <module>
    from sigenergy2mqtt.__main__ import main
  File "/root/.local/share/pipx/venvs/sigenergy2mqtt/lib/python3.12/site-packages/sigenergy2mqtt/__main__.py", line 1, in <module>
    from sigenergy2mqtt.config import Config
  File "/root/.local/share/pipx/venvs/sigenergy2mqtt/lib/python3.12/site-packages/sigenergy2mqtt/config/__init__.py", line 487, in <module>
    apply_cli_to_env(arg, override)
  File "/root/.local/share/pipx/venvs/sigenergy2mqtt/lib/python3.12/site-packages/sigenergy2mqtt/config/__init__.py", line 436, in apply_cli_to_env
    if value != os.environ[variable]:
                ~~~~~~~~~~^^^^^^^^^^
  File "<frozen os>", line 714, in __getitem__
KeyError: 'SIGENERGY2MQTT_HASS_ENABLED'
[07:54:25] WARNING: Halt add-on with exit code 1
```

seud0nym, 2025-08-02:
Well, that will teach me to add something just before publishing a release :-(

Very sorry about that. I have just published 2025.8.2-1 which should fix it.

cottinglea, 2025-08-02:
10:38:47] INFO:  -> Parameter: 1
[10:38:47] INFO:  -> Parameter: --modbus-readonly
[10:38:47] INFO:  -> Parameter: --no-metrics
2025-08-02 10:38:47,634 INFO  __init__:24 Release 2025.8.2
Traceback (most recent call last):
  File "/root/.local/bin/sigenergy2mqtt", line 3, in <module>
    from sigenergy2mqtt.__main__ import main
  File "/root/.local/share/pipx/venvs/sigenergy2mqtt/lib/python3.12/site-packages/sigenergy2mqtt/__main__.py", line 1, in <module>
    from sigenergy2mqtt.config import Config
  File "/root/.local/share/pipx/venvs/sigenergy2mqtt/lib/python3.12/site-packages/sigenergy2mqtt/config/__init__.py", line 487, in <module>
    apply_cli_to_env(arg, override)
  File "/root/.local/share/pipx/venvs/sigenergy2mqtt/lib/python3.12/site-packages/sigenergy2mqtt/config/__init__.py", line 436, in apply_cli_to_env
    if value != os.environ[variable]:
                ~~~~~~~~~~^^^^^^^^^^
  File "<frozen os>", line 714, in __getitem__
KeyError: 'SIGENERGY2MQTT_HASS_ENABLED'
[09:38:47] WARNING: Halt add-on with exit code 1
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped -

Problem still with me

boma304, 2025-08-02:
Hi, 
I can confirm that version 2025.8.2-1 fixed the problem for me. Thank you!

seud0nym, 2025-08-02:
Can you please try uninstalling and reinstalling the add-on? The first build picked up the old version, not the fixed version, and I had to do another build.

seud0nym, 2025-08-03:
> Problem still with me

@cottinglea : Did you try manually checking for updates in the add-on store? There was an issue with the first build of the amd64 version of the release, so I re-ran it and it should have been fixed.

In any event, I have had to publish version 2025.8.3 of the add-on today to fix _another_ bug, so try that one.

cottinglea, 2025-08-03:
Long day yesterday! Yes I deleted and reinstalled and can confirm all is working.

closed 2025-08-03 by seud0nym