seud0nym/sigenergy2mqtt #16: Home Assistant Addon - Error when publishing to PVOutput

justinfowler1980, 2025-08-04:
When attempting to publish to PVOutput the following log messages are printed in a loop causing system slowdowns etc.  

2025-08-04 21:40:11,365 WARNING service:83 PVOutputStatusService Removed key='v6' from payload because it is None
2025-08-04 21:40:11,365 ERROR status:61 PVOutputStatusService dictionary changed size during iteration

seud0nym, 2025-08-05:
I have made some changes that should resolve the 'dictionary changed size during iteration' and inifinite looping problem, but I have not been able to work out the underlying cause. Those errors are a consequence of bad error handling when the PVOutput service has not been updated with the Voltage states from the PV Strings. 

The thing is that I can't see _why_ it isn't being updated for those of you experiencing the issue. I have a 6kW Energy Controller which supports up to 2 PV strings, of which I am only using one. The state is updated in both Home Assistant and the PVOutput service without fail, and therefore I have not seen the error.

I have also made a change that will only display the 'not been updated' message once per hour, rather than on every status update.

I have just published Release 2025.8.5, which I sincerely hope fixes this issue.

justinfowler1980, 2025-08-05:
Further information - I have the 10 kW inverter (supporting 4 string), I use all 4 strings.   
Running in standalone docker this appears to be working fine, just fails when using the HA addon.

seud0nym, 2025-08-06:
> Maybe this could have to do with the problem?

No, it should deal correctly with the zero states from unused strings.

@justinf555 : Has the 2025.8.5 release fixed the issue?

justinf555, 2025-08-07:
The issue appears to be fixed now.   Thankyou.