seud0nym/sigenergy2mqtt #19: MQTT TLS support

mr-suw, 2025-08-09:
I miss the feature to connect to a mosquito TLS server. Maybe it is easy to add?

seud0nym, 2025-08-10:
Adding encrypted communication to the broker using TLS is not difficult, and I will add it to the next release.

Authentication using client certificates is more complicated and I probably won't do that.

seud0nym, 2025-08-11:
I have just published release 2025.8.11, which contains:
* A new command line option `--mqtt-tls` to enable encrypted communication
* A new environment variable `SIGENERGY2MQTT_MQTT_TLS` which can be set to `true` to enable encrypted communication
* The Home Assistant Add-On configuration GUI contains a new option to enable communicaiton over TLS

Please re-open this issue if you find any problems.

closed 2025-08-11 by seud0nym