seud0nym/sigenergy2mqtt #20: Incorrect smart port power

dyeadon, 2025-08-09:
It seems to display the incorrect value of the power of the smart port 1 by a factor off 100 and also displays units of kWh not kw

seud0nym, 2025-08-09:
Thanks for the report.

Yes, you are correct. I have obviously copied the consumption sensor and then not changed the unit and device class. 

It will be fixed in the next release.

Are you using the Home Assistant add-on? If so, you will get notification of repairs that need to be done to those sensors after you upgrade.

seud0nym, 2025-08-10:
A question for you: Does Smart Port power affect reported consumption? What I mean is, does the Home load in the mySigen app include the Smart Port consumption, and if it does, then I assume that Consumed Power in `sigenergy2mqtt` is incorrect by the value of your Smart Port consumption?

dyeadon, 2025-08-10:
> Thanks for the report.
>
> Yes, you are correct. I have obviously copied the consumption sensor and then not changed the unit and device class.
>
> It will be fixed in the next release.
>
> Are you using the [...]

I originally installed through HACs and then installed the integration, then installed the Addon after. I also had another setup from another person, so I had many integartions. I have since uninstalled all 3 versions then just installed the addon. I think the smart port is working ok now. although do you there are several different sensors that display similar information the for the hoem assistant energy configuration data. Do you know which ones should be used. I think the smart port and the total power are correct