seud0nym/sigenergy2mqtt #40: Inverter PowerFactor note reported any more due to (incorrect) sanity check

martin-ecatch, 2025-09-30:
This is the relevant warning in the log:
WARNING base:407 PowerFactor Publishing SKIPPED: Failed to get state (ValueError('Value 64539 is not within sanity check range 0 - 1000 (state=64539 previous_states=[])'))

In this case, power factor is 64.5% which is correct. But it's rejected by this sanity check.

Eventually, I get
WARNING base:422 PowerFactor Publish DISABLED until Tue Sep 30 22:08:33 2025 - MAX_FAILURES exceeded: 10

This is a new issue in release 2025.9.24 and I suspect it's related to this change: "Added sanity checks to inverter power factor and temperature sensors", this is the commit https://github.com/seud0nym/sigenergy2mqtt/commit/8e81248ba46d38178817f017de69716eebef216e and this is the offending line 1275 in the commit: https://github.com/seud0nym/sigenergy2mqtt/commit/8e81248ba46d38178817f017de69716eebef216e#diff-2921473462600c88747becabc6cd21cc9571f4a29c816ab9ed729a09e24d4a54R1275

Looks like the sanity checks are applied to the raw value, instead of the scaled one? Should this be set 100000?

martin-ecatch, 2025-10-01:
I see, you're right! The Power Factor is normally a value between 0 and 1. When I see 64, I assumed it was 64%.
This happens when the inverter is idle btw.
Thank you for reporting it to Sigenergy. I'm going to close this issue, because there's nothing to fix.

closed 2025-10-01 by martin-ecatch

seud0nym, 2025-11-19:
I have just released 2025.11.19 that contains a work-around for this Modbus error. If it gets an invalid value like 64536, it will recalculate the Power Factor manually from the Active Power and Reactive Power, and then publish the calculation, rather than just ignoring the invalid value.